robotics-in-concert/rocon_tools #81: [rocon_interactions] support video streamer dynamic binding

jihoonl, 2015-03-19:
Need parameter to configure `video_web_server` parameter to stream out video streaming for webapp

* __VIDEO_STREAMER_ADDRESS__
* __VIDEO_STREAMER_PORT__

stonier, 2015-03-20:
Does the web app have a rosbridge connection to the system already?

If so, then it can get any parameterisation itself right? Is this really a job of the interactions manager?

stonier, 2015-03-20:
I am guessing you're including this because you have multiple web app interactions requiring the same parameter and don't want to have this same parameter copy pasted in multiple locations.

We do for rosbridge because that supports an entire category of interactions. This would only support some special use cases and I think the concept breaks if you want to consider serving multiple streams anyway.

I'd rather keep things as minimal as possible rather than have an ever growing list of 'special variables'.

What is possibly a better solution is `parameter reflection`. Any parameter marked with magic characters in the interactions yaml should direct the interactions manager node to look up a parameter on the parameter server and serve that. I am not sure if it's better to just look up parameters in the private namespace, or have those magic character prefixed variables refer to a usual parameter lookup entity. 

e.g. for private lookup, perhaps `__video_streamer_adress__`

e.g. for namespace based `__ROS_PARAM__/video/streamer/address` or `__ROS_PARAM__~video_stream_address`

jihoonl, 2015-03-20:
:+1: __ROS_PARAM__ to give more flexibility sounds good.

jihoonl, 2015-04-02:
[image]

jihoonl, 2015-04-02:
Param name

* `__PARAM__` : param under namespace
* `__/PARAM__` : param in absolute path
* `__~PARAM__` : **NOT SUPPORTED**

commit 087b7f8a (2015-04-06): Merge pull request #83 from robotics-in-concert/param_binding

ros param binding to resolve #81
commit 555825bb (2015-04-06): ros param binding to resolve #81
closed 2015-04-06 by jihoonl