robotics-in-concert/rocon_tools #88: Google chrome launch error in web interaction demo

dhirajdhule, 2015-05-14:
Google chrome is not properly supported in ubuntu 14.04 with 64 bit machine. Following errors occur during chrome launch
Gtk-Message: Failed to load module "overlay-scrollbar"
Gtk-Message: Failed to load module "unity-gtk-module"
I googled and found that installing i386 versions of above packages on 64 bit machine is not possible and hence google chrome can not be used. 
Is there any way we can launch the interaction on mozilla or chromium?

jihoonl, 2015-05-15:
@dhirajdhule 

Can you tell how you started web interactions demo?

[rocon_remocon](https://github.com/robotics-in-concert/rocon_qt_gui/tree/indigo/rocon_remocon) is designed to choose available web browser to start web interactions.

In [this code](https://github.com/robotics-in-concert/rocon_qt_gui/blob/indigo/rocon_remocon/src/rocon_remocon/utils.py#L73-L87), it selects an available  web browser to start web interactions. Please remove google-chrome cleanly and try again. It should select chrominum-browser if google-chrome is not available.

jihoonl, 2015-05-15:
FYI,

I just added firefox support in rocon_remocon.

https://github.com/robotics-in-concert/rocon_qt_gui/issues/190

dhirajdhule, 2015-05-15:
Thanks,
It worked when I removed google chrome completely.
Now it is opening in chromium.

closed 2015-05-15 by dhirajdhule