robotics-in-concert/rocon_tools #93: standalone Web remocon, similar to android remocon app

dhirajdhule, 2015-06-05:
Can we have some webapp like android_remocon_app / rocon_remocon, which can give user access to all available interactions? current way of launching a browser from rocon_remocon makes it platform dependent, means you always need some rocon_remocon or android remocon like app to launch these interactions which generates the unpredictable URL like following

http://192.168.1.102:8080/trial%20.html?interaction_data=%7B%22remappings%22%3A+%7B%22%2Fchatter%22%3A+%22%2Fbabbler%22%7D%2C+%22display_name%22%3A+%22vallabh%22%2C+%22parameters%22%3A+%7B%22rosbridge_address%22%3A+%22192.168.1.102%22%2C+%22rosbridge_port%22%3A+9090%7D%7D 

I think, we can make it more user friendly if we could have an webapp as a remocon, where user can select rosbridge address and then select different webapps according to the interactions. 

I found this web app which i think exactly does the same role as above. If it does, then I was not able to use it.   http://webapp.robotconcert.org/rocon_web_remocon/
I had started web interactions demo on 11311 and then manually opened this app in browser, but it gives error when I put master URI as ws://localhost:11311

dhirajdhule, 2015-06-08:
sorry, was trying to connect on ros masters port instead of rosbridge port,
solved.

closed 2015-06-08 by dhirajdhule

jihoonl, 2015-06-08:
:neckbeard: