robotics-in-concert/rocon_tools #102: [rocon_interactions] more rapp watcher problems

stonier, 2015-08-12:
Crash in runtime:

```
  File "/opt/groot/rocon_ws/src/rocon_tools/rocon_interactions/src/rocon_interactions/rapp_watcher.py", line 248, in run
    list_rapps_service_names = rocon_python_comms.find_service('rocon_app_manager_msgs/GetRappList', timeout=rospy.rostime.Duration(60.0), unique=False)
  File "/opt/groot/rocon_ws/src/rocon_tools/rocon_python_comms/src/rocon_python_comms/services.py", line 112, in find_service
    next_service_type = get_service_headers(service_name, service_uri).get('type', None)
  File "/opt/ros/indigo/lib/python2.7/dist-packages/rosservice/__init__.py", line 124, in get_service_headers
    return rosgraph.network.read_ros_handshake_header(s, StringIO(), 2048)
  File "/opt/ros/indigo/lib/python2.7/dist-packages/rosgraph/network.py", line 365, in read_ros_handshake_header
    raise ROSHandshakeException("connection from sender terminated before handshake header received. %s bytes were received. Please check sender for additional details."%b.tell())
ROSHandshakeException: connection from sender terminated before handshake header received. 0 bytes were received. Please check sender for additional details.
```

stonier, 2015-08-12:
This was on the robot - not across a network. Nothing else crashed or halted. 

1. Needs to be resolved so it doesn't crash
2. find the root cause - I'd like to know why this happens as it may be indicative of something deeper
3. should at least produce a warning and be gathering with more relevant information until we know the root cause

asmodehn, 2015-08-12:
So :
```
File "/opt/groot/rocon_ws/src/rocon_tools/rocon_python_comms/src/rocon_python_comms/services.py", line 112, in find_service
    next_service_type = get_service_headers(service_name, service_uri).get('type', None)
```

I had a look and there is already that : 
```python
            try:
                next_service_type = get_service_headers(service_name, service_uri).get('type', None)
            except ROSHandshakeException:
                # happens when shutting down (should handle this somehow)
                continue
```

So I m guessing despite the warning, the process doesn't really die ? If true, that would already solve 1.

stonier, 2015-08-13:
Also noticed a pattern today - everytime this showed up:

1. nodelet bonds break
2. 3d short range sensor goes down

stonier, 2015-08-13:
Got it....`rosgraph.network.Exception`, not `roslib.network.Exception`.

stonier, 2015-08-13:
This looks like a case of just hammering the master. The `rapp watcher` is calling four separate `find_xxxx` apis per second, and if there is no service available, then these are looping at 10Hz themselves.

If we're going to have gateway and rostful banging the master already, I think it's time to drop handles to the rapp manager in a constant namespace locally and statically remap to these at launch time.

stonier, 2015-08-19:
Rapp watcher is moved out and `find_xxx` calls are no longer required by the interactions manager since the rapp manager now provides local (fixed) handles.

closed 2015-08-19 by stonier