robotics-in-concert/rocon_tools #116: missing OS targets vivid etc

tfoote, 2015-11-18:
I'm running on vivid and there are lots of warnings due to `vivid` not being in this file. Such as the following:

    [WARN] [WallTime: 1447805278.124815] Rapp Manager : 'rocon_apps/lion_chirp' is invalid [operating_system specification is invalid or not officially registered (see https://github.com/robotics-in-concert/rocon_tools/blob/indigo/rocon_uri/src/rocon_uri/rules/rules.yaml)[vivid]]

This file only seems that it can get out of date? Why reproduce the data here?

stonier, 2015-12-11:
Short term, I've added the operating system to the rules.

Long term an agnostic rapp like chirp shouldn't care about the operating system underneath. I think this has been fixed in our recent branch, but I'll leave this open here till I confirm.

Yes, it can get out of date. If I recall, this was so validition at various levels could be tighter than just string matching any pair of strings at runtime (e.g. `vivix` would get caught as an error earlier in development rather than at runtime later when failing against an `os` with `vivid`).  Tradeoff of course is it needs more maintenance. If I did this again, I'd be more inclined to avoid the maintenance, as I see you are.

What did you mean by reproduce? Is it getting produced elsewhere?

tfoote, 2015-12-11:
I meant reproduce the list of possible os's to validate against. You can probably pull it in from rospkg. https://github.com/ros-infrastructure/rospkg/blob/master/src/rospkg/os_detect.py

stonier, 2015-12-14:
Ah, got it. Tasks for this issue:

* [ ] rapps with no dependencies should be agnostic about a missing operating system match.
* [ ] operating system list should at least be initialised from rospkg