robotics-in-concert/rocon_tools #126: Wifi down, rosgraph exception in connections

stonier, 2016-03-30:
```
[ INFO] [1459241906.689244246]: Bond broken, exiting
Traceback (most recent call last):
  File "/opt/yujin/amd64/indigo-devel/lib/rocon_python_comms/connection_cache.py", line 23, in <module>
    conn_cache.spin()
  File "/opt/yujin/amd64/indigo-devel/lib/python2.7/dist-packages/rocon_python_comms/connections.py", line 482, in spin
    new_conns, lost_conns = self.conn_cache.update()
  File "/opt/yujin/amd64/indigo-devel/lib/python2.7/dist-packages/rocon_python_comms/connections.py", line 365, in update
    c.generate_type_info_msg()
  File "/opt/yujin/amd64/indigo-devel/lib/python2.7/dist-packages/rocon_python_comms/connections.py", line 177, in generate_type_info_msg
    self.type_msg = rosservice.get_service_type(self.name)
  File "/opt/yujin/amd64/indigo-devel/lib/python2.7/dist-packages/rosservice/__init__.py", line 147, in get_service_type
    return get_service_headers(service_name, service_uri).get('type', None)
  File "/opt/yujin/amd64/indigo-devel/lib/python2.7/dist-packages/rosservice/__init__.py", line 124, in get_service_headers
    return rosgraph.network.read_ros_handshake_header(s, StringIO(), 2048)
  File "/opt/yujin/amd64/indigo-devel/lib/python2.7/dist-packages/rosgraph/network.py", line 365, in read_ros_handshake_header
    raise ROSHandshakeException("connection from sender terminated before handshake header received. %s bytes were received. Please check sender for additional details."%b.tell())
rosgraph.network.ROSHandshakeException: connection from sender terminated before handshake header received. 0 bytes were received. Please check sender for additional details.
[gopher/sensor_3d_short_range/downsampler-53] process has finished cleanly
```

asmodehn, 2016-03-31:
Interesting, it looks like the exception raised when losing network is not handled. Probably should be in rosservice... There are probably other cases like that lying around.
In our case losing network is expected and nothing fatal.
I think it should probably be handled at a low level in ros to make everything more robust with intermittent connections... unless ROS people usely expect things to break when network breaks ? 
I ll dive in more shortly.

stonier, 2016-05-11:
I think a low level library like ROS is fine throwing exceptions which lets the user decide what they want to do. If they decide to cover it up, then that simple library has to start handling connection topology and all the myriad decisions that come with it.

At any rate, they won't handle it ROS1 for sure as they have no idea who might be using it and relying on those exceptions being thrown.